yorotsuya/otk-poc #2: Big size data write via NFC will collapse the process

yorotsuya, 2018-06-28:
**Describe the bug**
When writing data via NFC with size >= 950 bytes, it will collapse the process making the board enter zombie mode.

**To Reproduce**
Create a big text data and write through NFC.

**Expected behavior**
data size <= 947 bytes, OK
948 bytes <= data size < 950 bytes, Write Error but did not collapsed
data size >= 950 bytes, Write Error and collapsed

**Smartphone (please complete the following information):**
 - Device: Sony Xperia XA2
 - OS: Android 8.0
 - NFC Tools v6.5

**Additional context**
Should consider to implement a hardware reset function.

jengjy, 2018-07-04:
The bug doesn't happen on current version now.
It might be fixed on moving _ndefMsgBuf[] to global so that the callback function can access it w/o issue.

yorotsuya, 2018-07-12:
Tested with large data. Now OTK accepts data size up to 4091 bytes and return 'Write Error' normally for larger size data without blocking the process.

closed 2018-07-12 by yorotsuya