yorotsuya/otk-poc #4: Default NFC read data should containt public address instead of public key

yorotsuya, 2018-06-28:
Description:
Currently the default published field to NFC reader contains a public key with full information which can be used to calculate child node. For user to make RX transaction, only public address is needed. Thus, put public key as private field which is only revealed by request.

jengjy, 2018-07-04:
On current version, it exposes public key(example: 0386BB10901CB267BF2D7F134D4743B57FDF6B8A1EA774F7CADD9C1C5B18773E91) instead of serialized public key(example: xpub69B6kqSazv6wsJ4nRjc5pJams8WQSz9DTnbQ3R6tgAG...).

yorotsuya, 2018-07-22:
Issue fixed in commit 6a3f8d94700e31b48ff7564e759b5ca26bec456b, 
added BTC_publicKeyToAddr(), now NFC read the compressed bitcoin address.

closed 2018-07-22 by yorotsuya