yorotsuya/otk-poc #22: Add set transaction fee feature

yorotsuya, 2019-03-02:
Currently 'Make Payment' is applying default transaction fees which should be able to be configured by users.

jengjy, 2019-06-03:
So it's too add a field for the user to set tx fee in "Make Payment" page, right?
And by default, how much or how much percentage will the fees be?

yorotsuya, 2019-06-03:
TX fee is a dynamic number. It's used to be set as x satoshi per byte, so the total TX fee is the rate times TX data size.

As how the fee is decided, it is usually depending on how many transactions in bitcoin memory pool. When all transactions in the memory pool can all be included in the next block, a user can set as low rate as even zero which is still likely to be included in the new block, but when there are more transactions which are greater than the size of a block, a higher tx fee that a transaction pays will increase its priority in the memory pool and makes it faster to be included in the new blocks. The lower fee is set, the longer a TX has to wait or even get cancelled.

There is no strict rules in setting the TX fee. If everyone is not paying TX fee or paying the same fee, the transactions will most likely be included by the order as it's broadcasted. However, when bitcoin is in congestion, there are always people who are willing to pay a slight higher fee to save the confirmation time. So somebody created a fee calculator as a web service to give user advice rates in various priority.

從 ProtonMail 手機版送出

--------原始郵件--------

yorotsuya, 2019-06-03:
This feature should be a configurable preference of the app (so is the 'fee included' feature), and an optional function in making payments. A user should not be requested to set the value each time, but can always change it before completing the TX.

The options for user should be simple options like high, mid, low priority, and a customized rate that user can set manually.

從 ProtonMail 手機版送出

--------原始郵件--------

jengjy, 2019-06-03:
I see.

jengjy, 2019-06-07:
What's your suggestion on the value (sat/b) for High, Mid and Low level?

yorotsuya, 2019-06-08:
Check this out,

https://www.buybitcoinworldwide.com/fee-calculator/

I think we are using legacy transaction for blockcypher, I don't know if they support segwit or not. Default should be low for six blocks, mid for four blocks, and high for two blocks. However, the suggested fee from fee calculator is dynamic. We can put fix rate first. I think there is also http api offering fee suggestion.

Meanwhile, I checked my Coinomi wallet, it sets a fix rate of 50 sat/byte which seems to work just fine. I have not failed to send with that setting.

從 ProtonMail 手機版送出

--------原始郵件--------

jengjy, 2019-06-08:
Bad news.
Blockcypher provides two ways to set tx fees when generating new transaction. One is to set "preference" to "low/medium/high" and the value for low/medium/high is dynamic and decided by Blockcypher. They are average fee for tx to be confirmed within 7~/3~6/1~2. Another way is to see fee(in satoshi) for the transaction.  The bad news is I tried to set "preference" and it doesn't work. And for setting fee, it's not practical because I don't know the data size before I generate a new transaction...Set fee directly in satoshi without knowing tx data size doesn't make sense.

Another way to set tx fee is to generate tx by ourselves rather than by Blockcypher and calculate fee according to tx data size. I have to figure it out and try.

jengjy, 2019-06-08:
I tried to set "preference" over testnet, maybe main net work... I will give it a try.

yorotsuya, 2019-06-08:
Ok, thanks buddy, just keep me updated.

從 ProtonMail 手機版送出

--------原始郵件--------

jengjy, 2019-06-08:
No luck. Main net is the same. To support this feature, I have to do something manually. Let me study it...