yorotsuya/otk-poc #30: OTK-APP / Add option to authorize OTK with secure PIN instead of FP

yorotsuya, 2019-03-07:
Allow user to run OTK commands with secure PIN to pre-authorized and no need for fingerprint authorization.

jengjy, 2019-06-03:
Where should this feature locate? A button, such as "Pre-authorized with PIN", in "Make Payment" page?

yorotsuya, 2019-06-03:
I think the easiest and most convenient place is the system menu on the top right corner. The pre-authed function is kind of a standalone function which will prompt a dialog to allow a user to enter the pin code and send the command to a (any) otk.

I was thinking to have an improved UI design similar to WeChat interface. The first version of the implementation is mainly for demo and features/functions verification and test. We will soon be publish the app on Google play, so we will need a better user friendly interface. But I think the UI design can be a separate job from feature implementation. For feature implementation, we can just consider to insert a feature in the current interface which is most reasonable.

從 ProtonMail 手機版送出

--------原始郵件--------